capacitor-community/bluetooth-le #633: Android only shows "Unknown" device name when scanning with requestDevice()

mley, 2024-03-08:
**Describe the bug**
When using the `requestDevice()` method to scan for BLE heart-rate monitors, Android shows the devices only as "Unknown" with their MAC address. I test this with a Polar H9 and a Garmin watch in broadcast mode. On iOS the names of the devices are shown correctly.

```
await BleClient.requestDevice({
        services: ['0000180d-0000-1000-8000-00805f9b34fb'],
        scanMode: ScanMode.SCAN_MODE_LOW_LATENCY,
      });
```

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on BLE HR monitor
2. Call requestDevice()
3. See result

**Expected behavior**
I expect the pick device dialog to also show the name of the device

**Screenshots**
[image]

[image]

**Plugin version:**
- @capacitor-community/bluetooth-le: 3.1.2

**Smartphone (please complete the following information):**
- Device: Samsung Galaxy A22
- OS: Android 13
- Browser: Chrome

pwespi, 2024-04-06:
I'm unable to reproduce this on my test devices. Are you seeing this on multiple Android devices?

peitschie, 2024-04-06:
Worth noting, iOS has 2 mechanisms for reporting BLE name, while Android only uses the advertised name. I wonder if the above is only showing the GAP name in iOS, which is cached after connection?

See https://forums.developer.apple.com/forums/thread/19381

@mley capturing the specific advertising packet using nrf Connect would perhaps make it clearer what's being broadcast.

mley, 2024-04-08:
Thanks for the reply. 

I tried again today with the same Android device and the same Polar H9 and this time the name of the device came back correctly. It might have just been a hiccup, fixed by a reboot of the Android phone...

Sorry for the inconvenience. I will close the issue then.

closed 2024-04-08 by mley

lroal, 2025-10-19:
I still have this isse with v 7.2.0