capacitor-community/bluetooth-le #645: Trouble detecting specific BLE device when using requestLEScan with targeted services

urbilog, 2024-04-16:
**Describe the bug**
Description: I'm encountering an issue while using the requestLEScan function in the @capacitor-community/bluetooth-le plugin. When I attempt to scan for a specific BLE device by targeting its service UUIDs using the services option, I'm unable to detect anything. However, when I leave the services option as an empty array, I can detect the desired device in the list of discovered devices. Additionally, I confirmed that the correct UUID is present in the result.rawAdvertisement?.buffer.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the requestLEScan function with the services option targeting specific service UUIDs.
2. Attempt to detect a specific BLE device.
3. Notice that no devices are detected.
4. Remove the specific service UUID targeting and rescan.
5. Observe that the desired device is detected in the list of discovered devices.

**Expected behavior**
When specifying the service UUIDs in the services option, I expect the requestLEScan function to detect the targeted BLE device.

**Plugin version:**
- @capacitor-community/bluetooth-le: ["^3.0.2"]

**Desktop (please complete the following information):**
- OS: [MacOS]

**Smartphone (please complete the following information):**
- Device: [Oppo A94 5G]
- OS: [Android 13]

**Additional context**
Note: I've verified that the issue persists across multiple attempts and have reviewed the documentation for any potential misconfigurations. Any assistance or guidance on resolving this issue would be greatly appreciated.

eggbeard, 2024-05-20:
Still present in ^6.0.0

peitschie, 2024-05-20:
@urbilog are you able to confirm that the advertising packet has the requested service UUID in it?

In BLE, there is a distinction between the services the peripheral supports, and the services the peripheral advertises. What does the advertised services contain for one of these peripherals you're scanning for?

eggbeard, 2024-05-21:
So I have something like this:
```js
async BLEServiceTest(){
    await BleClient.initialize({ androidNeverForLocation: true });
    const isBonded = await BleClient.isBonded('48:A4:93:4B:19:B5');
    console.log('isBonded1', isBonded); //True if paired
    await BleClient.connect('48:A4:93:4B:19:B5');
    const services = await BleClient.getServices('48:A4:93:4B:19:B5'); //By Mac address
    console.log('Services of 48:A4:93:4B:19:B5 are:', services);

    // Of these services '38eb4a80-c570-11e3-9507-0002a5d5c51b' is the one I care about
    // This is a Zebra thermal printer
    // https://www.zebra.com/content/dam/zebra/software/en/application-notes/AppNote-BlueToothLE-v4.pdf
    const zebraDevice: BleDevice = await BleClient.requestDevice({
      services:['38eb4a80-c570-11e3-9507-0002a5d5c51b']
    });
    // Produces a long list that does include the device above
    console.log('BleClient.requestDevice with Zebra Service:', zebraDevice);
  }
'''

When connecting to the printer by mac address services are logged as:
![image](https://github.com/capacitor-community/bluetooth-le/assets/3766758/b8f8804b-86b3-4794-9cf0-ac6473446a98)

Note this includes '38eb4a80-c570-11e3-9507-0002a5d5c51b'. Is this a supported or an advertised service?

'''js
async BLEServiceTest2(){
    await BleClient.initialize({ androidNeverForLocation: true });
    
    const device: BleDevice = await BleClient.requestDevice();
    // Produces a long list that does include the device above
    console.log('BleClient.requestDevice with no options:', device);
  }
'''

This does present UI for a long list of devices, but this is not really useful for an end user since they don't now about device ids and uuids.  For example:

![Screenshot_20240521_082830_Volt - SAA Mobile](https://github.com/capacitor-community/bluetooth-le/assets/3766758/d76e9bdd-7bd4-446b-8767-9f80cbb72a7c)

If I have not already connected to the device then it does appear in this list, but it's often hard to find so really I want to filter to something more relevant.

So now I try something like:
'''js
const zebraDevice: BleDevice = await BleClient.requestDevice({
      services:['38eb4a80-c570-11e3-9507-0002a5d5c51b']
    });
'''

Giving me:
![Screenshot_20240521_085108_Volt - SAA Mobile](https://github.com/capacitor-community/bluetooth-le/assets/3766758/02ee2ab0-000c-450a-9c85-c5097f68d478)

![Screenshot_20240521_085125_Volt - SAA Mobile](https://github.com/capacitor-community/bluetooth-le/assets/3766758/a0fd4a93-d3a0-443d-87be-d7131f3b3e75)

Is this the wrong way to do this?

Finally if I try:
'''js

  await BleClient.initialize({ androidNeverForLocation: true });

  const devices: BleDevice[] = await BleClient.getConnectedDevices(['38eb4a80-c570-11e3-9507-0002a5d5c51b']);
  devices.forEach(device => {
    console.log('Device:', device);
  });
'''

that also returns nothing

![image](https://github.com/capacitor-community/bluetooth-le/assets/3766758/fe93fc55-c8e1-4a66-a526-a82317f39dde)

peitschie, 2024-05-22:
Hi @eggbeard 

You're showing the list of services **after** connection. But, the filtering during a scan applies based on the advertising payload only. Most peripherals advertise only a small subset of the services they support.

I'd suggest looking at your peripheral's advertising payload (**without** connecting) in [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) to see whether the service you want to filter by is included in the advertising packet.

eggbeard, 2024-05-22:
This is the answer.  I finally understand a little better. [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) is very useful in understanding what is going on.  It had not occurred to me that the advertising payload could be so different from the services list once connected.

pwespi, 2024-07-22:
Thank you @peitschie for your help in debugging this. Seems to be resolved now.

closed 2024-07-22 by pwespi