capacitor-community/bluetooth-le #655: requestLEScan: Max gatt client reached: 32, no response from capacitor

andrew-aladjev, 2024-05-20:
**Describe the bug**
Gatt leaks in Android BLE. It is well know thing, but capacitor ble plugin didn't provide a way to handle this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to use `requestLEScan` in android app.
2. Destroy app process during scan (don't call `stopLEScan`).
3. Repeat this thing 32 times.

```
Capacitor/Plugin: To native (Capacitor plugin): callbackId: 7257068, pluginId: BluetoothLe, methodName: addListener
Capacitor: callback: 7257068, pluginId: BluetoothLe, methodName: addListener, methodData: {"eventName":"onScanResult"}
Capacitor/Plugin: To native (Capacitor plugin): callbackId: 7257069, pluginId: BluetoothLe, methodName: requestLEScan
Capacitor: callback: 7257069, pluginId: BluetoothLe, methodName: requestLEScan, methodData: {"services":[],"allowDuplicates":true}
DeviceScanner: Start scanning.
BluetoothAdapter: isLeEnabled(): ON
bt_stack: [INFO:le_scanning_manager.cc(141)] RegisterScanner in shim layer
bluetooth: packages/modules/Bluetooth/system/gd/hci/le_scanning_manager.cc:597 register_scanner: Unable to register scanner, max client reached:32
BluetoothLeScanner: onScannerRegistered() - status=128 scannerId=0 mScannerId=0
Capacitor/Console: File: http://localhost/ - Line 232 - Msg: undefined
```

I wanted to handle this error and restart BLE, but capacitor is not capable to produce this error. I mean `requestLEScan` is not calling callbacks, is not throwing errors, it just hangs silently.

**Expected behavior**
`requestLEScan` throws some error.

**Plugin version:**
- @capacitor-community/bluetooth-le: [e.g. 3.1.4]

**Smartphone (please complete the following information):**
- Device: Google Pixel 7a API 33
- OS: Android 13

andrew-aladjev, 2024-05-21:
I have a good news: I can't reproduce this issue anymore in `v6.0.0`.

pwespi, 2024-07-22:
I'm closing as it doesn't seem to be reproducible in the latest version. Let me know if it happens again.

closed 2024-07-22 by pwespi