capacitor-community/bluetooth-le #661: Timeout exception and disconnect during BLE write for iOS

WillooWisp, 2024-06-25:
**Describe the bug**
Depending on version of BLE server (different versions of BlueZ), BLE write on same iOS device behaves different. The problem is that against some BLE servers the behavior is incorrect, where we always gets a timeout exception, even though the data is actually written successfully. Increasing the timeout for iOS does not make any difference, except that it takes longer for the timeout to occur, i.e. exception to get caught. It works as expected using Android against the same BLE server and BlueZ version.

Seems a bit related to #659, regarding timeout and device disconnect that we also get when writing.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a BLE write against BlueZ 5.55, e.g. on Debian 11, from an iOS device.
2. Timeout exception occurs even though write is successful.

**Expected behavior**
A successful BLE write without timeout exception.

**Plugin version:**
- @capacitor-community/bluetooth-le:6.0.0

**Smartphone (please complete the following information):**
- Device: iPhone 8
- OS: iOS16.7.8

WillooWisp, 2024-06-28:
The workaround for now, is to make the connection stateless, i.e. connect before each read and disconnect right after, same for write. That way read and write works without timeouts almost all the time, and since we do not keep the connection open, we also do not get disconnected when timeout occurs. This works okay for us, since we don't use BLE notifications, but there must be some bug in the framework related to iOS and timeouts.

pwespi, 2024-07-22:
Do you know if this behavior is specifically happening when connecting to a device with BlueZ version 5.55 or any BlueZ version?

closed 2024-08-22 by github-actions[bot]

WillooWisp, 2024-08-22:
> Do you know if this behavior is specifically happening when connecting to a device with BlueZ version 5.55 or any BlueZ version?

It for sure happens often with 5.55 but I guess it happens for any version below 5.60, but the problem is that these newer versions of BlueZ are only available for Debian 12 and above. It works perfectly fine on Android with any version however.