capacitor-community/bluetooth-le #662: Sometimes the scan does not return localName and device.name but only device.deviceId

emanueletoffolon, 2024-07-05:
**Describe the bug**
Sometimes the scan does not return localName and device.name but only device.deviceId

**To Reproduce**
Start the scan using requestLEScan with the default options

**Smartphone (please complete the following information):**
- Device: some Android phone
- OS: Android
- Version: any

**Plugin version:**
- @capacitor-community/bluetooth-le:6.0.0

Has this ever happened to you?

Thanks

pwespi, 2024-07-22:
This is expected because not all devices advertise a name. 

Do you have a peripheral where you expect a name and don't get one?

emanueletoffolon, 2024-07-22:
Yes, sometimes the same device returns the name and deviceId and sometimes just the deviceId.  I noticed this by keeping track of the deviceId since on Android it corresponds to the Mac address and never changes unlike on iOS.  I tried to do several scanning tests with nRF connect and I didn't encounter this problem, so I decided to write here to ask if anyone has had the same problem.

Thanks

bonatoc, 2024-09-13:
@emanueletoffolon I have the same problem under Android.
Sometimes it's the ID of the devices, sometimes it's the name (as per BLE GAP).