capacitor-community/bluetooth-le #663: [Android] Android 14 / api 34 Crashing issue on android 14 devices

Marcello17, 2024-07-12:
**Describe the bug**
I've upgraded my app using :
    compileSdkVersion = 34
    targetSdkVersion = 34
    
    and it gave this error:
    FATAL EXCEPTION: main
    
java.lang.RuntimeException: Unable to start activity ComponentInfo{mypackagename/MainActivity}: java.lang.SecurityException: mypackagenamei: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts

I see the files show this:
  context.registerReceiver(stateReceiver, intentFilter)
  does this need to be adjusted to accommodate android 14?

**To Reproduce**
Steps to reproduce the behavior:
Set the API to 34:

   compileSdkVersion = 34
    targetSdkVersion = 34

**Screenshots**
[image]

**Plugin version:**
- @capacitor-community/bluetooth-le: 6.0.0 or 3.1.1 [ Tried on both]

**Smartphone (please complete the following information):**
- Device: Samsung S21 / S24
- OS: Android 14

peitschie, 2024-07-13:
Just copying over my notes from https://github.com/don/cordova-plugin-ble-central/issues/1020#issuecomment-2226819996

Relevant documentation from Android on this:

* https://developer.android.com/develop/background-work/background-tasks/broadcasts#context-registered-receivers
* https://developer.android.com/about/versions/14/behavior-changes-14

Although these are all system broadcasts, because they're being registered via an Activity, they're effectively context-registered receivers, and thus need a `RECEIVER_EXPORTED` flag, as the app needs intents coming from outside the application itself (i.e., from the system).

lajith111, 2024-07-17:
Same issue happen on cordova capacitor. Please check  https://github.com/cordova-sms/cordova-sms-plugin/pull/233/files
Now i dont have android 14 device to check
```
i am not sure in kotlin  if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.TIRAMISU)
{
   context.registerReceiver(bondStateReceiver, intentFilter,Context.RECEIVER_EXPORTED)
}
else
{
  context.registerReceiver(bondStateReceiver, intentFilter)
}
```

peitschie, 2024-07-17:
@Marcello17 or @lajith111 do either you have the capacity to see if https://github.com/capacitor-community/bluetooth-le/pull/667 fixes the problem for you?

closed 2024-07-17 by Marcello17

pwespi, 2024-07-22:
Thank you @Marcello17 for reporting back.

I can confirm that with 

```
compileSdkVersion = 34
targetSdkVersion = 34
```
my test app works fine on an Android 14 device.