capacitor-community/bluetooth-le #668: [Android] Error "Device is not connected" on Write o Read any characteristic

chriscastro94, 2024-07-19:
**Detailed Description of the Issue**
_Issue Summary:_
After successfully connecting to a BLE device using the @capacitor-community/bluetooth-le plugin in an Ionic 7 Angular application, attempting to read or write characteristics returns an error stating "device is not connected." This issue occurs even though the initial connection appears to be successful.

**Steps to Reproduce:**

1. Initialize the BLE client with BleClient.initialize().
2. Request the BLE device using BleClient.requestDevice() with appropriate service UUIDs.
3. Disconnect to the device using BleClient.disconnect(device.deviceId).
4. Connect to the device using BleClient.connect(device.deviceId).
5. Attempt to read or write a characteristic using BleClient.read(device.deviceId, serviceUUID, characteristicUUID) or BleClient.write(device.deviceId, serviceUUID, characteristicUUID, value).

**Observed Behavior:**

The device connects successfully, as confirmed by the connection logs.
Upon attempting to read or write a characteristic, an error is thrown indicating that the device is not connected.
Expected Behavior:

After a successful connection, reading from or writing to a characteristic should succeed without errors.
Additional Context:

The issue seems to be intermittent, as the connection is initially established but subsequent read/write operations fail.
Adding delays and retry logic does not resolve the issue.
This issue occurs on multiple Android devices and versions.
Environment:

Ionic Version: 7
Capacitor Version: 6.0.0
BLE Plugin Version:    "@capacitor-community/bluetooth-le": "^6.0.0"
 Pls Help

pwespi, 2024-07-22:
A disconnect can also be triggered by the peripheral. When connecting you can pass an optional onDisconnect callback, see https://github.com/capacitor-community/bluetooth-le?tab=readme-ov-file#connect. Is this callback called between the successful connection and your read or write attempt?

chriscastro94, 2024-07-22:
I noticed that the disconnection event only triggers after a write or read attempt. However, by changing the autoconnect parameter to true, the connection works consistently.
[image]