capacitor-community/bluetooth-le #674: Connectivity Issue in IOS APP

madhan-SEET, 2024-07-26:
**Describe the bug**
I try to connect the ble device using ios app to make connection I am using the BlieClient.connect(deviceID) 
Note: Same code working fine android app 
I checked my Xcode logs 
BluetoothLe - Connected to peripheral. Waiting for services discovery.
BluetoothLe - didDiscoverServices
BluetoothLe - didDiscover CharacterisiticFor 1 1 
BluetoothLe - Resolve onDisconnected

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: [e.g. 0.4.0]

**Smartphone (please complete the following information):**
- Device: [e.g. iPhone12]

**Additional context**
Add any other context about the problem here.

peitschie, 2024-07-30:
Hi @madhan-SEET 

Out of interest, did you end up figuring out this issue?

madhan-SEET, 2024-07-31:
Yes, I updated the plugin now it's working now.

-- 
*Disclaimer: The content of this email is confidential and intended for the 
recipient specified in message only. It is strictly forbidden to share any 
part of this message and its attachments with any third party, without a 
written consent of the sender. If you received this message by mistake, 
please reply to this message and follow with its deletion. Please note that 
as the recipient, it is your responsibility to check the email for 
malicious software. The integrity and security of this email cannot be 
guaranteed over the Internet. Therefore, the sender will not be held liable 
for any damage caused by the message.
*

peitschie, 2024-07-31:
Awesome to hear! Thanks for sharing your solution 😃