capacitor-community/bluetooth-le #683: characteristic doesn't support read operation, but it does :/

eried, 2024-09-02:
Hi! thanks for this add-on, I am trying to connect to a simple ESP32 device that exposes a characteristic as:

```c
  // Create the characteristic
  BLECharacteristic* pCharacteristic = pService->createCharacteristic(
    CHARACTERISTIC_UUID,
    BLECharacteristic::PROPERTY_READ | 
    BLECharacteristic::PROPERTY_NOTIFY | 
    BLECharacteristic::PROPERTY_INDICATE);
```

After connecting to the device (this part works ok), when I try to get if I can read the value:

```ts

      const result = await BleClient.read(device.deviceId, this.SERVICE_UUID, this.CHARACTERISTIC_UUID);
```

I always get:

`app.component.ts:141 Error fetching fish data: NotSupportedError: GATT operation not permitted.`

﻿Even when I can read it on another app on the same PC:

[image]

I am missing something very obvious?

peitschie, 2024-09-04:
That looks like it ought to work. Have you tried [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) on mobile (there's an Apple version too) just to confirm it's not a BLE issue?

nrf Connect sometimes can show some deeper diagnostics information too, to help root-cause what's going on there.

eried, 2024-09-04:
Hi, thanks. It is very similar, also can read in nrfConnect

[image]

On the web there is no permissions I could have missed right?, I am still puzzled on why it does not work :/

peitschie, 2024-09-10:
Just clarifying @eried ... which platform ar eyou running on? Is this Web (via Chrome?), Android or iOS?

eried, 2024-09-10:
Hi, web via chrome, windows 10

peitschie, 2025-04-13:
I've just realised that this is probably connected with #704, which really just stresses that the Web Bluetooth platforms are not always working that reliably 😒 

Unfortunately, I don't have any great insights here.

closed 2025-04-14 by eried