capacitor-community/bluetooth-le #685: feat : Broadcast Signal

bootexe, 2024-09-04:
**Is your feature request related to a problem? Please describe.**
I would like to build a chat app in bluetooth but its seems I can't broadcast a signal that can be identified by the same app on other phones
**Describe the solution you'd like**
Two method one to broadcast a signal and and other to receive a broadcast signal with a callback as parameter (or event handler) 
im not a expert but I think those two won't be possible in web so iam only expecting this in iOS and Android 

**Describe alternatives you've considered**
I've search the documentation of every other plugin and aim alone to build this app so cant build it in Java and Swift to access this feature in a native way

**Additional context**
no additional context

peitschie, 2024-09-04:
Hi @bootexe 

What you're asking for requires both a BLE central implementation (which this plugin can help with) and a BLE peripheral implementation, which will need a different plugin to support:

* https://github.com/randdusing/cordova-plugin-bluetoothle
* https://github.com/don/cordova-plugin-ble-peripheral

But... it's still a lot of work after those two pieces are in play.

Fundamentally, the feature you're asking for here is far too large to fit into this type of plugin.

bootexe, 2024-09-05:
Hi ! thanks for your response
its leading me somewhere where I can pursue my research to build this app
thanks again !

Petar-Pankov, 2024-10-04:
@bootexe did you find any solution? I am looking for something similar for iOS. I just need to broadcast and ID from device 'A', and device 'B' to see this ID.