capacitor-community/bluetooth-le #697: Throw error codes instead of message strings in the exception

wm-eisos, 2024-10-18:
**Is your feature request related to a problem? Please describe.**
So far the current exceptions thrown have string based messages which is sometimes event differenet between the platforms, whiuch would make it hard to interpret the resulting issue. such as in (https://github.com/capacitor-community/bluetooth-le/blob/83e2b1b45c65ad0835210826814830ae9d575ea8/ios/Plugin/DeviceManager.swift#L50) for iOS and (https://github.com/capacitor-community/bluetooth-le/blob/83e2b1b45c65ad0835210826814830ae9d575ea8/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/BluetoothLe.kt#L139) for Android which basically are permession denials however they are different strings which means the parser on the app side would need to take care about bopth scenarios and this owuld be way easier if it was error codes and would prevent the issue of someone changing the string inside the library causing the parser on the appp end to fail

**Describe the solution you'd like**
Have error codes instead of string based exceptions so that the user can interpret the values of the error codes way easier than whole strings.

pwespi, 2024-10-20:
Thank you for this feature request. It's a good idea.

The `reject` method of `PluginCall` actually supports adding an error code (as string) in addition to an error message, so this could be implemented in a non-breaking way.

Android:
https://github.com/ionic-team/capacitor/blob/cb325a32501118a6ac77071341e1a40f47300866/android/capacitor/src/main/java/com/getcapacitor/PluginCall.java#L74-L75

iOS:
https://github.com/ionic-team/capacitor/blob/cb325a32501118a6ac77071341e1a40f47300866/ios/Capacitor/Capacitor/CAPPluginCall.swift#L45-L46

Would you be able to prepare a PR for this?

wm-eisos, 2024-10-21:
Hey,

Unfortuanatly im not that expecrienced in capacitor plugnis, i believe it would take me longer to implement it. As well as the face that i dont have a MacOS device whiche means iOS development is not possible for me.

However i would also mention that if there is an enum mapping the errors to states it would be nice as i currently saw that the error code as a string.

Best Regards
Wajdi