capacitor-community/bluetooth-le #704: Web Bluetooth (Chrome Ubuntu) Scanning

vicatcu, 2024-10-24:
**Describe the bug**
I get a permission prompt when i initiate a Bluetooth scan using `BleClient.requestLEScan` but 

**To Reproduce**
Steps to reproduce the behavior:
Do all the pre-requisite enablement of experimental features in chrome://flags as shown here:
[image]

Enable Bluetooth in the operating system settings and observe that scan results are apparent in the system dialog.

Set a breakpoint in Chrome Dev Tools in the callback function for:
```
BleClient.requestLEScan(
   {services: []}, 
   (result) =>  {
      console.log(result); // set breakpoint here
   }
)` 
```
... and observe that it never executes, despite the same code working fine on iOS and Android builds.

The following prompt appears in Chrome:
[image]

... but clicking "Allow" doesn't make a difference, and Chrome doesn't seem to "remember" that I clicked allow either - it asks every time.

**Expected behavior**
Invoking a Bluetooth scan should cause the callback to fire in Web Bluetooth same as on iOS and Android. Allow dialog should not come up every time a scan is invoked?

**Screenshots**
N/A

**Plugin version:**
- @capacitor-community/bluetooth-le: 3.1.4

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04.5 LTS
- Browser: Google Chrome
- Version: 130.0.6723.69 (Official Build) (64-bit) 

**Additional context**
   NodeJS : v18.20.2
   npm    : 10.5.0
   OS     : Linux 6.9
   Angular: 16.2.9
   @ionic/angular@8.3.1
   @capacitor/cli: 5.7.5
   @capacitor/core: 5.7.8

pwespi, 2024-10-28:
Note that `requestLEScan` of Web Bluetooth is still experimental and has bugs. On the [implementation status](https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md#scanning-api) page Linux is not even listed as supported. Therefore I don't think there is anything that can be done about this in the plugin.

You could try to use `navigator.bluetooth.requestLEScan` directly and I suspect you would get the same results.

JooYoo, 2025-04-08:
I also have the same issue. 
Does it mean scanning Ble devices on the Web is not possible with this plugin? Since it's stuck on the "Block, Allow" page.

JooYoo, 2025-04-08:
> I think it's a buggy chrome implementation... if you copy the requestLEScan directly in the DevTools console it works for some reason, but when running from javascript code it doesn't work

You are right. I pasted 
```js
navigator.bluetooth.requestDevice({
  acceptAllDevices: true
}).then(device => {
  console.log('Found:', device);
});
```
In Chrome inspector console, it works as expected.