capacitor-community/bluetooth-le #710: Unable to read Data from bluetotth sesnor on 100ms

safiUllahRai, 2024-11-04:
A clear and concise description of what the bug is.
I am developing the application where it use this plugin to interact with Bluetooth pressure sensor called "M5600" where I have to fetch pressure data from sensor. Application is fetching pressure data on 1 second interval even app is fetching data on 500ms but it is giving error when I am setting the interval value to 100ms

**Screenshots**
[image]
[image]

I have attached the 2 screenshots where it is fetching data at 500ms and giving error at 100ms interval

**Plugin version:**
- @capacitor-community/bluetooth-le: ^6.0.0

peitschie, 2024-11-05:
Hi @safiUllahRai 

Can you show what the characteristic read code on the javascript side looks like?

Sometimes a peripheral will have it's own limitations that can prevent reading data too quickly.

peitschie, 2024-11-06:
In your screenshot above... it appears that sometimes you end up with a new `read` request while the previous one is still not yet returned?

[image]

How are you triggering these reads currently? Do you wait for the last read to complete first, or are you triggering these on a `setInterval` or similarly fixed timer?

Usually, overlapping reads on the same characteristic like this are not well supported.