capacitor-community/bluetooth-le #711: Plugin doesn't work in iOS 18

gassssty, 2024-11-04:
Hi guys,

I'm trying to add this plugin for use on iOS phones but I'm not having any success in implementing it. However, for Android phones, it works perfectly.
Actually, the code is the same for both platforms. Let me explain a little bit what's happening with images and functions in code:
- In our app we have a button where the initialize() is done:

```
initializeBleClient(): Promise<boolean> {
    return new Promise(async (resolve, reject) => {
      try {
        await BleClient.initialize();
        resolve(true);
      } catch (error) {
        console.error(error);
        reject(false);
      }
    })
  }
```

- Once initialized, it does this function (scan the devices):
```
scanDevicesByOperatorUUID(): Promise<BleDevice> {
    return new Promise(async (resolve, reject) => {
      try {
        const selectedDevice = await BleClient.requestDevice(
          {
            services: [this._serviceUUID]
          }
        );
        console.log("Selected device: ", selectedDevice);        
        this.setSelectedDevice(selectedDevice);
        resolve(selectedDevice);
      } catch (error) {
        reject(undefined)
      }
    });
  }
```
[image]

- We select the device shown in the list:
```
connectToPickio(): Promise<boolean> {
    return new Promise(async (resolve) => {
      try {
        if (this.getSelectedDevice()) {
          await BleClient.disconnect(this.getSelectedDevice()!.deviceId);
          BleClient.connect(this.getSelectedDevice()!.deviceId, () => this.disconnectActions()).then(() => { 
            this.trackDeviceRSSI();
            this.signalModalShowing = 0;  
            resolve(true);
          }).catch(() => resolve(false));
        } else {
          resolve(false);
        }
      } catch (error) {
        resolve(false);
      }
    });
  }
```

- It is connecting at the operating system level and not the application level, making it impossible for us to connect and move forward:
[image]

And then, we have some logs that we saw in the Xcode App:
⚡️  To Native ->  BluetoothLe removeListener 51121821
⚡️  To Native ->  BluetoothLe addListener 51121822
⚡️  To Native ->  BluetoothLe connect 51121823
⚡️  BluetoothLe - Connecting to peripheral <CBPeripheral: 0x3018109c0, identifier = 338676A7-6DF4-02F6-210B-A84762F07FA6, name = pickio-004-005, mtu = 517, state = connected>
⚡️  [error] - ERROR {"errorMessage":"Connection timeout"}
⚡️  BluetoothLe - Connected to device <CBPeripheral: 0x3018109c0, identifier = 338676A7-6DF4-02F6-210B-A84762F07FA6, name = pickio-004-005, mtu = 517, state = connected>
⚡️  BluetoothLe - Resolve connect|338676A7-6DF4-02F6-210B-A84762F07FA6 Successfully connected.
⚡️  BluetoothLe - Connected to peripheral. Waiting for service discovery.
⚡️  BluetoothLe - Reject connect Connection timeout
ERROR MESSAGE:  {"message":"Connection timeout","errorMessage":"Connection timeout"}
⚡️  [error] - {"message":"Connection timeout","errorMessage":"Connection timeout"}

[image]

iOS Version: 18.1
Angular: 18.0.0
Ionic: 8.0.0
Capacitor Community BLE: 6.0.0

Thank you very much, guys!! :D

peitschie, 2024-11-05:
Hi @gassssty 

Did this app work with previous versions of iOS?

I have found that iOS tends to be more timing sensitive than Android. Given this is timing out in the service discovery phase, I wonder if perhaps the device you're connecting to is taking too long to finish discovery.

Do you have any control over the peripheral itself? Do you know anything about it's connection interval, connection latency etc? 
Some of these details can be discovered with the Android version of [nrf Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp), and can be helpful here.

gassssty, 2024-11-05:
Hi!

No, it doesn't work on an older version, specifically 17.4. Actually on Android it always works on the first try and there's nothing wrong with the code, at least from what it seems.

**Do you have any control over the peripheral itself?**
Yes, it is a computer.

About the peripheral information, here you have:
- Device Type -> Classic and LE
- Advertising Interval -> 1288 ms.
- Advertising Type -> Legacy

And we don't have more information in the app nrf Connect. FYI: in Android is never connecting as OS level and with iOS always, with different iOS versions.

Thanks!!

peitschie, 2024-11-06:
Interesting.

That advertising interval is very slow. Do you have any details about the connection interval or latency? (I believe nrf Connect on Android can show this information too).

How many services/characteristics are exposed via this peripheral?

gassssty, 2024-11-07:
Well, there isn't more information than this. And we have 22 characteristics.

peitschie, 2024-11-07:
22 characteristics is quite a lot!

Can you try adding a very long timeout to your `connect` call? E.g.,
```
await BleClient.connect(device.deviceId, (d) => console.log("disconnected"), { timeout: 120000 });
```

With nrf Connect, it's a little bit tricky, but if you swipe to the right on the bottom left of the connected screen... you'll see this console slide out with some additional debug info:

[image]

gassssty, 2024-11-07:
Ahh great, so we are going to try to reduce this characteristics. Which number is the maximum (or recommendable) to put in?

Is true that with 2 characteristics in iOS 17 we can connect but never with 22. And then in iOS 18, not even with 2 or 22 characteristics... but it's weird that in Android works fine with 22.

Finally... thanks to say that because we don't know how to use the app properly hahaha. Here you have a screenshot of the latency and interval: 
[image]

peitschie, 2024-11-07:
> Is true that with 2 characteristics in iOS 17 we can connect but never with 22. And then in iOS 18, not even with 2 or 22 characteristics... but it's weird that in Android works fine with 22.

This is strange. On Android, that discovery is taking only about 500ms... I wonder why it's taking so much longer on iOS.
The default iOS timeout is 10s.. were you able to test this with a very long timeout? (e.g., 60s or more)?

I can't answer for sure what the upper limit is. I've definitely used things with several services and 5-8 characteristics per service. Never 22 on a single service... but you're saying that this isn't working with 2 on iOS?

There are some subtle differences between iOS and Android's connection paths that I've occasionally hit where a peripheral connects on Android (usually) and not iOS. I can't really guess the cause in your case though. Do you get any debug logs from the PC side of the connection, to spot any differences?

gassssty, 2024-11-13:
Yes, we tried with 120000 ms in the timeout option and it did not success... but we are going to try to reduce this 22 characteristics to 2 or something like that.

Exactly, is not working in 2 differente iOS devices, with different versions.

In theory in Android Studio it shows the same logs as the XCode does. The strange 'thing' is that in iOS is connecting always as SO and never in Android.

israeloriente, 2025-02-26:
I'm having the same issue. Any solution for the timeout problem on iOS?

matak, 2025-03-21:
the problem is that you can not call 

await BleClient.initialize()

twice, bcs when you call later than start scanning, it lost reference to scanned object and the bl manager never connect to device, just call initialize at the very first start of your script, just once, not more

gassssty, 2025-03-21:
I know @matak, but is what I am saying at the beginning of the post. I tried a lot of things and it still doesn't work. I don't know... 5 months later is something fixed in iOS?

matak, 2025-03-21:
> I know [@matak](https://github.com/matak), but is what I am saying at the beginning of the post. I tried a lot of things and it still doesn't work. I don't know... 5 months later is something fixed in iOS?

so, you are sure, that your lifecycle doesnt call the initialize twice? for example REACT.strictmode etc., reload page etc. bcs the problem is that native code lost reference to scanned device, if you before connection again call scanning it will work