capacitor-community/bluetooth-le #713: BleClient.writeWithoutResponse undefined

tpRamakrishna, 2024-11-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

able to connect to the device 
BleClient.writeWithoutResponse is giving a response as undefined.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: [e.g. 0.4.0]

**Desktop (please complete the following information):**
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**
- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

peitschie, 2024-11-11:
Hi @tpRamakrishna 

What response are you expecting? Are you able to post a code snippet demonstrating the problem?

tpRamakrishna, 2024-11-11:
const battery = await BleClient.read(this.deviceId, this.serviceId, this.characteristicUuid);
    console.log('battery level', battery.getUint8(0));
    const boldOn = new Uint8Array([27, 69, 1]);
     await BleClient.writeWithoutResponse(this.deviceId, this.serviceId, this.characteristicUuid, new DataView(boldOn.buffer));
  
let array = new Uint8Array([0xCC, 0x24, 0x33]);
    await BleClient.writeWithoutResponse(this.deviceId, this.serviceId.toLowerCase(), this.characteristicUuid.toLowerCase(), new DataView(array.buffer));

I can connect to the device and get the notifications from the device and also getting response onse also but visually print is not happening from the printer.

peitschie, 2024-11-12:
Unfortunately, this is going to be related to the commands you need to send to the Bluetooth printer itself. This isn't an area I can really help with, as it's not really about Bluetooth itself.

You'll need to look at the printer's SDK or interface documentation to figure out the right structures to send.

Having said that, if you do find a case where the plugin does not send the right information, do feel free to re-open this ticket.

closed 2024-11-12 by peitschie