capacitor-community/bluetooth-le #714: getConnectedDevices does not return bonded connected devices

emanueletoffolon, 2024-11-11:
On some Android devices, if the bonded device is connected, the getConnectedDevices method does not return it.
This does not occur systematically on all Android devices, but I have encountered it only on some.
For example, I had no issues on the Google Pixel 8a, but I did on a Motorola Moto G71.

Steps to reproduce the behavior:
1. Go to 'Connected Devices' on Android Setting
2. Click on 'Pair new device' and do it
3. Exec getConnectedDevices
4. No results may be returned

**Expected behavior**
I would expect to receive connected devices, even if bonded

**Screenshots**

> getConnectedDevices on Motorola Moto G71

[image]

**Plugin version:**
- @capacitor-community/bluetooth-le:6.0.2

**Smartphone:**
- Device: Motorola Moto G71 and Google Pixel 8a
- OS: Android 12 (Moto G71) and Android 15 (Pixel 8a)

**Additional context**
I tried using Cordova Bluetooth LE Plugin and connected bonded devices are returned without any issues even on Moto G71, but I would like to avoid using this plugin.

> getConnectedDevices on Motorola Moto G71 with Cordova Bluetooth LE Plugin

[image]

Let me know if any further information is needed to resolve the issue

peitschie, 2024-11-11:
Hi @emanueletoffolon 

I just wanted to clarify exactly which is the cordova plugin you're referring to here.

Do you mean https://github.com/randdusing/cordova-plugin-bluetoothle or https://github.com/don/cordova-plugin-ble-central?

emanueletoffolon, 2024-11-12:
Hi @peitschie, 
I'm referring to https://github.com/randdusing/cordova-plugin-bluetoothle.

Thanks

peitschie, 2024-11-12:
And the API call? Is this https://github.com/randdusing/cordova-plugin-bluetoothle?tab=readme-ov-file#retrieveconnected ?

peitschie, 2024-11-12:
Digging into the code, it seems that cordova-plugin-bluetoothle uses Android's [getBondedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices()) method.

https://github.com/randdusing/cordova-plugin-bluetoothle/blob/212bdc870868d51ae47de2210936a1f19b22f7d1/src/android/BluetoothLePlugin.java#L1348

I agree this functionality is not present in this plugin.

A PR implementing a new Android-only `getBondedDevices` method would be welcome (rather than changing the behaviour of the existing `getConnectedDevices` method).

emanueletoffolon, 2024-11-12:
Yes, in my app I wrote an interface to allow the user to choose which plugin to use.
Below I report the methods I wrote to retrieve the connected devices.

### Capacitor
```
  async getConnectedDevices(): Promise<BleDeviceInfo[]> {
    let services: string[] = [];
    if (this.api.devicePlatform == 'ios'){
      services = BleServices.filters.optionalServices;
    }
    return  await BleClient.getConnectedDevices(services);
  }
```

### Cordova
```
  async getConnectedDevices(): Promise<BleDeviceInfo[]> {
    const devices = await this.bluetoothLe.retrieveConnected({services: BleServices.filters.cordova});
    let ret: BleDeviceInfo[] = [];
    let item: BleDeviceInfo = {deviceId: ""};
    for(const i in devices){
      item = {
        deviceId: devices[i].address, name: devices[i].name
      };
      ret.push(item);
    }
    return ret;
  }
```

emanueletoffolon, 2024-11-12:
> Digging into the code, it seems that cordova-plugin-bluetoothle uses Android's [getBondedDevices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#getBondedDevices()) [...]

Thanks for the quick support. I'm not very familiar with Github. Do I have to open the PR myself?

peitschie, 2024-11-12:
> Thanks for the quick support. I'm not very familiar with Github. Do I have to open the PR myself?

Yes. GitHub has a lot of documentation about how to contribute code back to a repository 🙂

emanueletoffolon, 2024-11-12:
Ok, I'll try. Thanks for all

peitschie, 2024-11-21:
This is fixed via the new API @emanueletoffolon was kind enough to add: https://github.com/capacitor-community/bluetooth-le?tab=readme-ov-file#getbondeddevices

Feature is released in 6.1.0

closed 2024-11-21 by peitschie