capacitor-community/bluetooth-le #717: perf: improve byte to hex string conversion optimizations (#701)

paolosanchi, 2024-11-13:
**Describe the bug**
this commit breaks startNotification() on js side:
perf: improve byte to hex string conversion optimizations (https://github.com/capacitor-community/bluetooth-le/pull/701)

The issue is caused by bytesToString(), because it returns a hex string without spaces between the hex couple of values.

**To Reproduce**
Steps to reproduce the behavior:

```
val value = bytesToString(data)
val originalData = stringToBytes(value)
```

originalData is different from data

This is what I have debugging
[image]

```
data: [77, 83, 71, 95, 73, 78, 73, 84, 95, 79, 75, 58, 48, 10]
originalData: [77]
value: "4D53475F494E49545F4F4B3A300A"
```

**Expected behavior**
Value shoud be like this instead:
`"4D 53 47 5F 49 4E 49 54 5F 4F 4B 3A 30 0A"
`

the issue resides in here:
https://github.com/capacitor-community/bluetooth-le/blob/8e99165de13a72d5e0d45b1f3985ac96d7c5bd93/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Conversion.kt#L21

because the following function from the kotlin framework StringsJVM.kt returns the string without spaces:

```
public actual fun CharArray.concatToString(): String {
    return String(this)
}
```

**Screenshots**
see above

**Plugin version:**
- @capacitor-community/bluetooth-le:6.0.2

**Desktop:**
not applicable

**Smartphone:**
Device: Huawei MediaPad 5 Lite
OS: Android 8
Version: SDK 26

peitschie, 2024-11-13:
Hi @paolosanchi 

Just to confirm, are you finding this change in internal behaviour is causing an observable difference when used as part of `BleClient.startNotifications`?

paolosanchi, 2024-11-14:
I've found this issue because the callback of startNotification() on the js side receives always an empty string.
Indeed, the following function expect be bytes in the string to be separated by a space
https://github.com/capacitor-community/bluetooth-le/blob/8e99165de13a72d5e0d45b1f3985ac96d7c5bd93/src/conversion.ts#L19

peitschie, 2024-11-14:
The code here is a bit different to what you're describing 🙂 

The real flow here is:

1. `startNotification` receives a hex string here: https://github.com/capacitor-community/bluetooth-le/blob/main/src/bleClient.ts#L670
2. It calls `this.convertValue`, which forwards on to `hexStringDataToView`: https://github.com/capacitor-community/bluetooth-le/blob/main/src/bleClient.ts#L706
3. `hexStringDataToView` was carefully refactored to process a non-space separated hex string: https://github.com/capacitor-community/bluetooth-le/blob/main/src/conversion.ts#L44

I've just tested this again locally, and can confirm that for me at least, the proper values are coming through:

[image]

[image]

As you can see... the `hexStringToDataView` is correctly handling the space-less hex-encoded bye array.

Are you able to confirm the conversion.ts implementation embedded in your app matches this code, i.e., there's not a stale version of the plugin source somewhere in your setup?

Can you share an example of a full payload you're transferring, so I can see if there's any issues with the size or data contained?

paolosanchi, 2024-11-14:
@peitschie my bad, you are right. I've got this issue because I use ionic's Live Update, and I had the version 6.0.2 for android and 6.0.1 for web (I forgot to commit the package.json).

It is fine if the old version is not compatible with the new one, the opposite would not be good, wich this is not the case.

When reporting the js code reference, I've been confused by textToDataView() wich uses .split(' '), but it is not used anymore. The code I've found in debug was relative to the old version, and I simply was not aware of it.

That said, the issue on the android side may still cause unwanted behaviour since stringToBytes() cannot handle strings without spaces, and stringToBytes().
However, it seems to be used only in write(), and it is the user that set directly the uid. I mean, it doesn't seem to be a regression.

My issue was not a bug, and it is solved on my side. If you think it doesn't worth to fix stringToBytes() I guess it's fine.
Sorry for the false alarm and thank you for the prompt reply.

closed 2024-11-15 by peitschie