capacitor-community/bluetooth-le #728: Connection time out

dmsuh, 2024-12-01:
**Describe the bug**
Logcat
```java
2024-12-01 08:50:05.813 28382-28483 DeviceScanner           org.ble.test                         D  Stop scanning.
2024-12-01 08:50:05.814 28382-28483 BluetoothAdapter        org.ble.test                         D  isLeEnabled(): ON
2024-12-01 08:50:05.814 28382-28483 BluetoothLeScanner      org.ble.test                         D  could not find callback wrapper
2024-12-01 08:50:05.815 28382-28483 DeviceScanner           org.ble.test                         D  Start scanning.
2024-12-01 08:50:05.816 28382-28483 BluetoothAdapter        org.ble.test                         D  isLeEnabled(): ON
2024-12-01 08:50:05.816 28382-28483 CheckPermission         org.ble.test                         E  _bluetooth code = 10 
2024-12-01 08:50:05.823 28382-28560 BluetoothLeScanner      org.ble.test                         D  onScannerRegistered() - status=0 scannerId=6 mScannerId=0
2024-12-01 08:50:05.833 28382-28382 Capacitor/Console       org.ble.test                         I  File:  - Line 353 - Msg: undefined
2024-12-01 08:50:05.872 28382-28382 Capacitor/BluetoothLe   org.ble.test                         V  Notifying listeners for event onScanResult
2024-12-01 08:50:09.201 28382-28382 ViewRootImpl            org.ble.test                         D  Bottom Down mCompactScale : 1.125 needScale() : false downY : 1476 downX : 63
2024-12-01 08:50:10.580 28382-28382 ViewRootImpl            org.ble.test                         D  Bottom Down mCompactScale : 1.125 needScale() : false downY : 1305 downX : 529
2024-12-01 08:50:10.674 28382-28382 Capacitor/Console       org.ble.test                         I  File: https://10.240.20.107:9500/src/pages/IndexPage.vue?t=1733032199711 - Line 60 - Msg: 50:54:7B:59:83:4A
2024-12-01 08:50:10.843 28382-28382 Capacitor/Plugin        org.ble.test                         V  To native (Capacitor plugin): callbackId: 66512855, pluginId: BluetoothLe, methodName: removeListener
2024-12-01 08:50:10.843 28382-28382 Capacitor               org.ble.test                         V  callback: 66512855, pluginId: BluetoothLe, methodName: removeListener, methodData: {"eventName":"onScanResult","callbackId":"66512853"}
2024-12-01 08:50:10.845 28382-28382 Capacitor/Plugin        org.ble.test                         V  To native (Capacitor plugin): callbackId: 66512856, pluginId: BluetoothLe, methodName: stopLEScan
2024-12-01 08:50:10.845 28382-28382 Capacitor               org.ble.test                         V  callback: 66512856, pluginId: BluetoothLe, methodName: stopLEScan, methodData: {}
2024-12-01 08:50:10.845 28382-28483 DeviceScanner           org.ble.test                         D  Stop scanning.
2024-12-01 08:50:10.847 28382-28483 BluetoothAdapter        org.ble.test                         D  isLeEnabled(): ON
2024-12-01 08:50:10.862 28382-28382 Capacitor/Console       org.ble.test                         I  File:  - Line 353 - Msg: undefined
2024-12-01 08:50:12.688 28382-28382 Capacitor/Plugin        org.ble.test                         V  To native (Capacitor plugin): callbackId: 66512857, pluginId: BluetoothLe, methodName: disconnect
2024-12-01 08:50:12.689 28382-28382 Capacitor               org.ble.test                         V  callback: 66512857, pluginId: BluetoothLe, methodName: disconnect, methodData: {"deviceId":"50:54:7B:59:83:4A"}
2024-12-01 08:50:12.691 28382-28483 BluetoothGatt           org.ble.test                         D  cancelOpen() - device: 50:54:7B:59:83:4A
2024-12-01 08:50:17.694 28382-28382 Device                  org.ble.test                         D  reject: disconnect Disconnection timeout.
2024-12-01 08:50:17.699 28382-28382 Capacitor               org.ble.test                         D  Sending plugin error: {"save":false,"callbackId":"66512857","pluginId":"BluetoothLe","methodName":"disconnect","success":false,"error":{"message":"Disconnection timeout."}}
2024-12-01 08:50:17.722 28382-28382 Capacitor/Console       org.ble.test                         I  File: https://10.240.20.107:9500/src/pages/IndexPage.vue?t=1733032199711 - Line 67 - Msg: Error: Disconnection timeout.
```

js code + html (quasar framefork)
```js
<template>
  <q-page class="row items-center justify-evenly">
    <example-component
      title="Example component"
      active
      :todos="todos"
      :meta="meta"
    ></example-component>
    <div :key="item.device.deviceId" v-for="item in modem">
    <div>
      {{item.device.deviceId}}
      <q-btn @click="connect(item.device.deviceId)">
        Подключится

      </q-btn>
    </div></div>
  </q-page>
</template>

<script setup lang="ts">
  import {onMounted, reactive, ref} from 'vue';
import type { Todo, Meta } from 'components/models';
import ExampleComponent from 'components/ExampleComponent.vue';
  import {BleClient, BleService, numberToUUID, ScanResult} from "@capacitor-community/bluetooth-le";

const todos = ref<Todo[]>([
  {
    id: 1,
    content: 'ct1'
  },
  {
    id: 2,
    content: 'ct2'
  },
  {
    id: 3,
    content: 'ct3'
  },
  {
    id: 4,
    content: 'ct4'
  },
  {
    id: 5,
    content: 'ct5'
  }
]);

const meta = ref<Meta>({
  totalCount: 1200
});
const modem = ref(<ScanResult[]>[])
  const delay = ms => new Promise(res => setTimeout(res, ms));
const scan= async (): Promise<void>=>{
  try {
    //const isAvailable = await Geolocation.checkPermissions();
    try {
      await BleClient.initialize({androidNeverForLocation: true});
      await BleClient.requestLEScan({namePrefix:"GL"}, (result) => {
        console.log(result);
        modem.value.push(result)
      });
      await setTimeout(async () => {
        await BleClient.stopLEScan();
      }, 5000);
    } catch (err: any) {
      console.log(err);
    }
  }catch (e) {
    console.log(e)
  }
};
onMounted(async ()=>{
  await scan()
})
  const postServices = reactive({ uuidW:'', uuidN:''})
  const connect = async (deviceId) => {
      try {
        console.log(deviceId)
        await delay(2000)
        await BleClient.disconnect(deviceId);
        await BleClient.connect(deviceId,(e=>console.log(e)),{timeout:9000});
        console.log('------------------------------')
        await delay(2000)
        /*const services:BleService[] =await BleClient.getServices(deviceId);
        for (let i=0; i<services.length; i++) {
          if (numberToUUID(0xffe0)===services[i].uuid){
            if (services[i].characteristics.length>1){
              for (let j=0; j<services[i].characteristics.length;j++){
                if(services[i].characteristics[j].properties.notify){
                  postServices.uuidN=services[i].characteristics[j].uuid
                }
                if(services[i].characteristics[j].properties.writeWithoutResponse){
                  postServices.uuidW=services[i].characteristics[j].uuid
                }
              }
            }else {
              if(services[i].characteristics[0].properties.notify){
                postServices.uuidN=services[i].characteristics[0].uuid
              }
              if(services[i].characteristics[0].properties.writeWithoutResponse){
                postServices.uuidW=services[i].characteristics[0].uuid
              }
            }
          }
        }*/
      } catch (e) {
        console.log(e)
      }
  };
</script>
```

**To Reproduce**
Steps to reproduce the behavior:
1.connect any device with this code

   

**Plugin version:**
 "@capacitor/android": "^6.0.0",
    "@capacitor/app": "^6.0.0",
    "@capacitor/cli": "^6.0.0",
    "@capacitor/core": "^6.0.0",
    "@capacitor-community/bluetooth-le": "^6.1.0",
    "@jcesarmobile/ssl-skip": "^0.4.0"

**Desktop (please complete the following information):**
-Android 9 api 28
- Browser [chrome]

**Smartphone (please complete the following information):**
- Device: [Oppo CPH1941]
- Android 9 api 28
- Browser [chrome]

**Additional context**
Add any other context about the problem here.

dmsuh, 2024-12-01:
[image]
if delete its, error not disappear

dmsuh, 2024-12-01:
I am confused by the error code 10[image]

dmsuh, 2024-12-01:
Android manifect here
```xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:tools="http://schemas.android.com/tools"
  xmlns:android="http://schemas.android.com/apk/res/android">

    <application
        android:allowBackup="true"
        android:icon="@mipmap/ic_launcher"
        android:label="@string/app_name"
        android:roundIcon="@mipmap/ic_launcher_round"
        android:supportsRtl="true"
        android:theme="@style/AppTheme">

        <activity
            android:configChanges="orientation|keyboardHidden|keyboard|screenSize|locale|smallestScreenSize|screenLayout|uiMode"
            android:name=".MainActivity"
            android:label="@string/title_activity_main"
            android:theme="@style/AppTheme.NoActionBarLaunch"
            android:launchMode="singleTask"
            android:exported="true">

            <intent-filter>
                <action android:name="android.intent.action.MAIN" />
                <category android:name="android.intent.category.LAUNCHER" />
            </intent-filter>

        </activity>

        <provider
            android:name="androidx.core.content.FileProvider"
            android:authorities="${applicationId}.fileprovider"
            android:exported="false"
            android:grantUriPermissions="true">
            <meta-data
                android:name="android.support.FILE_PROVIDER_PATHS"
                android:resource="@xml/file_paths"></meta-data>
        </provider>
    </application>

    <!-- Permissions -->
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />
  <uses-permission android:name="android.permission.BLUETOOTH_SCAN"
  android:usesPermissionFlags="neverForLocation"
  tools:targetApi="s" />

    <uses-permission android:name="android.permission.INTERNET" />
</manifest>
```

peitschie, 2025-01-04:
Hi @dmsuh 

I can't find anything obviously wrong with your code snippets here.

Are you able to see if this behaviour is the same across other Android devices as well?

Have you tried with a much larger connection timeout? E.g., 60seconds rather than 9seconds?