capacitor-community/bluetooth-le #741: After adding bluetooth-le to my Ionic project "npx cap run android" fails.

JulianStogger, 2025-01-09:
**Describe the bug**
After adding bluetooth-le to my Ionic project "npx cap run android" fails.
After removing bluetooth-le it works agian. 
I try it with JDK 21 and JDK 23

**OUTPUT:**
[Ionic] Running...
> npx cap run android --target=RFCRB03QD8R 
√ Copying web assets from www to android\app\src\main\assets\public in 1.13s
√ Creating capacitor.config.json in android\app\src\main\assets in 614.50μs
[info] Inlining sourcemaps
√ copy android in 1.19s
√ Updating Android plugins in 4.16ms
[info] Found 7 Capacitor plugins for android:
       @capacitor-community/bluetooth-le@6.1.0
       @capacitor-community/sqlite@6.0.2
       @capacitor/app@6.0.2
       @capacitor/haptics@6.0.2
       @capacitor/keyboard@6.0.3
       @capacitor/splash-screen@6.0.3
       @capacitor/status-bar@6.0.2
√ update android in 109.41ms
× Running Gradle build - failed!
[error] 
        FAILURE: Build failed with an exception.
        
        * Where:
        Settings file 'C:\Users\Julian\Desktop\JumpControl\jump-control\android\settings.gradle' line: 5
        
        * What went wrong:
        A problem occurred evaluating settings 'android'.
        > Could not open dsl generic class cache for script 'C:\Users\Julian\Desktop\JumpControl\jump-control\android\capacitor.settings.gradle' (C:\Users\Julian\.gradle\caches\8.2.1\scripts\51yolcau3fzvmekmvdv21a2k9).
        > BUG! exception in phase 'semantic analysis' in source unit '_BuildScript_' Unsupported class file major version 67
        
        * Try:
        > Run with --stacktrace option to get the stack trace.
        > Run with --info or --debug option to get more log output.
        > Run with --scan to get full insights.
        > Get more help at https://help.gradle.org.
        
        BUILD FAILED in 767ms
        
[error] npx cap run android --target=RFCRB03QD8R  Failed
[error] Android Failed.

**Expected behavior**
I expect the app to run.

**Plugin version:**
       @capacitor-community/bluetooth-le@6.1.0
       @capacitor-community/sqlite@6.0.2
       @capacitor/app@6.0.2
       @capacitor/haptics@6.0.2
       @capacitor/keyboard@6.0.3
       @capacitor/splash-screen@6.0.3
       @capacitor/status-bar@6.0.2

**Variables.gradle:**
    minSdkVersion = 22
    compileSdkVersion = 34
    targetSdkVersion = 34
    androidxActivityVersion = '1.8.0'
    androidxAppCompatVersion = '1.6.1'
    androidxCoordinatorLayoutVersion = '1.2.0'
    androidxCoreVersion = '1.12.0'
    androidxFragmentVersion = '1.6.2'
    coreSplashScreenVersion = '1.0.1'
    androidxWebkitVersion = '1.9.0'
    junitVersion = '4.13.2'
    androidxJunitVersion = '1.1.5'
    androidxEspressoCoreVersion = '3.5.1'
    cordovaAndroidVersion = '10.1.1'

peitschie, 2025-01-11:
Hi @JulianStogger 

To confirm, are you saying that if you remove just the bluetooth-le plugin, the app builds ok. It only fails after adding this plugin?

What version of Capacitor are you using here?

peitschie, 2025-01-11:
A quick google for the listed error message

>     > Could not open dsl generic class cache for script 'C:\Users\Julian\Desktop\JumpControl\jump-control\android\capacitor.settings.gradle' (C:\Users\Julian\.gradle\caches\8.2.1\scripts\51yolcau3fzvmekmvdv21a2k9).

Suggests it's caused by having multiple versions of JAVA available to the gradle build tools. I would suggest confirming that your `JAVA_HOME` environment variable is always correctly pointing to the JDK version you intend to be used, and that the `PATH` points to the correct bin folder for java/gradle/etc.

closed 2025-01-17 by JulianStogger