capacitor-community/bluetooth-le #748: ERROR {"errorMessage":"Connection timeout"} on connect iOS

israeloriente, 2025-02-27:
**Describe the bug**
I'm having issues connecting to my BLE device. Using the same code, it works on Android, but on iOS, it times out.

**To Reproduce**
Steps to reproduce the behavior:
1. await BleClient.connect(
        this.cubeUUID,
        (deviceId) => {
          console.log(`Disconnected .... ${deviceId}`);
        },
        { timeout: 30000 }
      );

**Plugin version:**
- @capacitor-community/bluetooth-le: ^6.1.0

**System:**
- OS: MacOS Sequoia
- Version 15.3.1

**Smartphone:**
- Device: iphone 11
- OS: iOS
- Version: 18.4

**Additional context**
[image]

peitschie, 2025-02-27:
Hi @israeloriente 

Are you able to try with another application on iOS (e.g., [nrf Connect](https://apps.apple.com/us/app/nrf-connect-for-mobile/id1054362403) is my normal go-to here) to see if you get similar symptoms?

Most of the time when I've seen this, is related to the peripheral itself, rather than anything we can influence on the phone side.

israeloriente, 2025-02-28:
Unfortunately, I don't have the equipment on hand right now.
When I do, I'll test it on nRF Connect, but as far as I know, this functionality used to work normally in the app I'm working on.

I'll get back to this chat when I have the peripheral 😁

carlossk, 2025-05-09:
I’m facing the same issue. In my case, I’m using a barcode scanner. I’m still researching how the plugin handles the timeout on iOS. I increased the timeout value and the scanner is disconnected automatically until the timeout completes.
in this example I increased the timeout 900000 = 15 minutes that the sacanner remains connected and then the scanner disconnect automatically.
`BleClient.connect(deviceId, deviceId => {
               //
            }, {
                timeout: 900000
            })`

israeloriente, 2025-06-21:
The problem was in my code. Honestly, I don’t remember exactly how I solved it, but basically, I rewrote all the logic from scratch, and it worked on this same version 6.1.0.

closed 2025-06-21 by israeloriente