capacitor-community/bluetooth-le #755: WriteWithResponse Method

marcelowolfsmartindustries, 2025-05-05:
**Is your feature request related to a problem? Please describe.**
Currently, the plugin only exposes the write method, which uses WriteWithoutResponse. However, some Bluetooth LE devices require WriteWithResponse to communicate properly. Without support for this, it's impossible to interact with those devices, which limits the app's compatibility.

**Describe the solution you'd like**
I would like the plugin to include an additional method, such as writeWithResponse(...), or alternatively, allow specifying the write type (withResponse or withoutResponse) as an option in the existing write method. This would provide more flexibility and compatibility with a wider range of BLE devices.

**Describe alternatives you've considered**
- Modifying the plugin directly to support WriteWithResponse, but this makes it harder to keep up with future updates via NPM.
- Using a different BLE plugin, but integration with Capacitor is often less complete or stable compared to this community plugin.

**Additional context**
WriteWithResponse is required for communication with certain devices that expect a write acknowledgment, such as medical sensors or industrial equipment.

pwespi, 2025-05-05:
The plugin exposes two write methods:
- `BleClient.write(...)`
- `BleClient.writeWithoutResponse(...)`

which correspond to the two write types you are mentioning. The first one is the "WriteWithResponse".

closed 2025-05-05 by pwespi

marcelowolfsmartindustries, 2025-05-10:
> The plugin exposes two write methods:
> 
> * `BleClient.write(...)`
> * `BleClient.writeWithoutResponse(...)`
> 
> which correspond to the two write types you are mentioning. The first one is the "WriteWithResponse".

Both are a Promise<void>. 

BleClient.write(write(deviceId: string, service: string, characteristic: string, value: DataView, options?: TimeoutOptions | undefined) => Promise<void>

BleClient.writeWithoutResponse(deviceId: string, service: string, characteristic: string, value: DataView, options?: TimeoutOptions | undefined) => Promise<void>

Maybe I'm misunderstanding. When I send a write command, the reply is sent as a notification?

peitschie, 2025-05-12:
@marcelowolfsmartindustries the Promise part here for `writeWithoutResponse` is so that you can wait until the OS has written the value out to the GATT. There is no implicit or built-in response assumed from the peripheral.

Basically, the promise will resolve when the call has filtered all the way down to either of these two places, depending on your OS:
* [iOS](https://github.com/capacitor-community/bluetooth-le/blob/main/ios/Plugin/Device.swift#L270)
* [Android](https://github.com/capacitor-community/bluetooth-le/blob/main/android/src/main/java/com/capacitorjs/community/plugins/bluetoothle/Device.kt#L200)

Neither of these wait for any feedback from the peripheral. You would await the promise in your code to catch any write failures that can still occur however (e.g., peripheral not connected, characteristic doesn't exist, some other kind of GATT error)