capacitor-community/bluetooth-le #757: Add Web Bluetooth Manufacturer Specific Data from requestDevice

Stevie-Ray, 2025-05-06:
**Is your feature request related to a problem? Please describe.**
I want to connect a Bluetooth Scale that uses Manufacturer Specific Data: I got it working in Web by setting: `optionalManufacturerData` in `requestDevice` and call `watchAdvertisements`.

https://googlechrome.github.io/samples/web-bluetooth/watch-advertisements-async-await.html

See: https://github.com/WebBluetoothCG/web-bluetooth/issues/598 it's also in the spec: https://webbluetoothcg.github.io/web-bluetooth/#dom-requestdeviceoptions-optionalmanufacturerdata

> The origin is also allowed to access any manufacturer data from manufacturer codes defined in options.optionalManufacturerData from the device’s advertisement data.

Would it be possible to implement `optionalManufacturerData` in `requestDevice` because the Scanning API / requestLEScan gives no result on Web.

if `optionalManufacturerData` is set we could add the following to bluethooth-le's `requestDevice`:

```
if (typeof device.watchAdvertisements === "function") {
  device.watchAdvertisements()
}
```
and check it by:

`navigator.bluetooth.addEventListener('advertisementreceived', this.onAdvertisementReceivedCallback);`

Thanks in advance!

Stevie

Stevie-Ray, 2025-05-07:
now i just noticed #752 by @steinerjakob

peitschie, 2025-05-07:
Support for this has just been released in 7.1.0 thanks to @steinerjakob 's hard work!

If you experience any issues, please feel free to raise an issue or PR 🙂

closed 2025-05-07 by peitschie

peitschie, 2025-05-07:
The use-case you're looking at above is supported via requestLEScan with the new manufacturer filters that #752 adds support for.

In the scan result, you can retrieve the manufacturer data here: https://github.com/capacitor-community/bluetooth-le/tree/main?tab=readme-ov-file#scanresult

An example is something like this:

```javascript
const companyIdentifier = 0x9988;
const scanOptions = {
  allowDuplicates: true,
  manufacturerData: [
    { companyIdentifier: companyIdentifier, dataPrefix: Uint8Array.from([0xff]) },
  ],
};

await ble.requestLEScan(scanOptions, (scan) =>
  console.log(
    "Manufacturer payload: ",
    scan.manufacturerData[companyIdentifier].getUint8(1)
  )
);

```