capacitor-community/bluetooth-le #761: Android AutoConnect functionality

juanaza, 2025-05-14:
**Is your feature request related to a problem? Please describe.**
The problem is that on a device disconnection, you can't really scan a BLE device when the phone is on background. Therefore could we get the autoConnect functionality that Android is offering?

**Describe alternatives you've considered**
This is included in capawesome-team/capacitor-bluetooth-low-energy but the plugin is giving me issues on old Andoid devices.

peitschie, 2025-05-14:
Surprisingly, the existing plugin **does** quite happily remain scanning and operational in the background. What is likely to be causing you problems however is the pause that occurs 5 minutes after the application is put into the background, due to Android's web view optimizations.

To keep your javascript/app code executing beyond this pause, you need to use a plugin such as [capacitor-background-mode](https://github.com/heffthedev/capacitor-background-mode), specifically the [disableWebViewOptimizations()](https://github.com/heffthedev/capacitor-background-mode?tab=readme-ov-file#disablewebviewoptimizations) feature.

DISCLAIMER: Web view optimisations are important for most applications to ensure they don't exceed Android's power budgets and end up black-listed. If you disable them, you need to take great care to ensure any animations and unnecessary global timers are paused when the app is put in the background, otherwise you will find your app continues to use CPU and eventually will be terminated by Android.

This is not the complete solution though for robust, background scanning and connectivity. I've outlined the depth of those challenges in https://github.com/capacitor-community/bluetooth-le/issues/643#issuecomment-2446345584

In short, I don't believe there's a gap in the current plugin here. Happy to hear alternative thoughts however...

juanaza, 2025-05-15:
Thanks for the reply, the thing is that the scanning in background mode is not even working just after the screen is locked. 5 minutes are not passing at all.

To summarize the workflow is the following.
- Foreground State
- Phone scans and connects to BLE device
- Phone subscribes to BLE device and receives data
- Background State (after phone is locked)
- Communication continues correctly through BLE subscriptions
- A sudden disconnection takes place (not even 1 minute since the phone got locked and running on background)
- Every 15 seconds after the phone disconnected from the BLE device i try to manually connect to the BLE device again but performing a scan to make sure the BLE device is available to connect. This last step is never executing judging by the Android Studio logs

Do you know what could be going on?

peitschie, 2025-05-16:
@juanaza interesting. Thanks for the extra detail there.

A quick question from your workflow here, does the MAC address of the device change between connections? Or does the device ID you use for connect remain consistent?

If the device ID remains constant, you shouldn't need a scan here as you can simply call `connect` to try again. If the device is not available, you'd eventually get a connection timeout, and can simply retry calling `connect` again and again.

On a separate note, when you are calling scan, what filters are you passing into the call here? Is it a fully unfiltered scan, or are you scanning for specific services or device names?

peitschie, 2025-05-19:
Hi @juanaza 

Does the scan keep functioning if you start it before with the services filter before the screen is turned off? I.e., start the scan, turn the screen off, and then see if the device successfully connects?

I'm trying to figure out if the scan is being prevented from starting, or if even an ongoing scan stops immediately.

Also, how long are you leaving the scan run for each time in the background?

peitschie, 2025-05-20:
The other point of consideration is whether you potentially need Android's `ACCESS_BACKGROUND_LOCATION` permission given your use-case: https://developer.android.com/develop/sensors-and-location/location/background

juanaza, 2025-05-20:
Hi @peitschie yes exactly I have tested that and if the scan starts before the screen is turned off, the reconnect functionality will take place correctly. As the scan gets called in the foreground and finalizes in the background.

I think the issue is the scan is being prevented from starting when it is called from the background.

I always scan for only 1500ms which is enough to get find the device I need

peitschie, 2025-05-20:
This sounds very much like we need that `ACCESS_BACKGROUND_LOCATION` permission for the scan.

For the interim, I'd suggest adding some code in the Android portion of your capacitor app to request this runtime permission (ChatGPT and friends should be able to help you out with the exact code snippet), and see if this allows the scan to be started while the device screen is off.

peitschie, 2025-05-20:
> I always scan for only 1500ms which is enough to get find the device I need

What do you do if no device is found within that scan? Do you sleep for a while before attempting another scan?

juanaza, 2025-05-20:
> > I always scan for only 1500ms which is enough to get find the device I need
> 
> What do you do if no device is found within that scan? Do you sleep for a while before attempting another scan?

Scan can be triggered in two ways:
1) Manually, if no devices are found an error message is shown to allow the user scanning again. If devices are found a picker is available to select one
2) Automatically, as part of the reconnection process, once the device disconnected every 15 seconds I will scan for 1500ms if the previously connected device is not found, the app will not try to reconnect and we won't scan until those 15 seconds have passed.

juanaza, 2025-05-20:
> This sounds very much like we need that `ACCESS_BACKGROUND_LOCATION` permission for the scan.
>
> For the interim, I'd suggest adding some code in the Android portion of your capacitor app to request [...]

will this eventually be fixed in the community plugin?

peitschie, 2025-05-20:
I'd definitely consider it if we can prove it works.