capacitor-community/bluetooth-le #762: Bluetooth filtering fails when only name is provided

JasonAsmk, 2025-05-15:
**Describe the bug**
When searching for a bluetooth device, for example with LE scan, and you provide only a name the filters that are built are null. That means that methods like `requestLEScan` which rely on filtering return instead all devices in the room. Filtering with name was working on 7.0.0 and broke in 7.1.0 as the bug was introduced with this commit https://github.com/capacitor-community/bluetooth-le/commit/c56b6709c42133d56ff7a6b96fdd8fed16253b12#diff-fa06ef5683f36c9d38e45547bec4d87909568138f7deb8f31b5a833c6217d146L826 . You can see on line 826 in Bluetooth.kt that there was a check for only name filtering that was removed.

If there was a reasoning behind this, I don't think it's documented.

**To Reproduce**
1. Init
2. Do a requestLEScan only with of the device provided, no manufacturer or services.
3. Returns all devices with no filtering.

**Expected behavior**
Only the devices that match the name should return.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Plugin version:**
- @capacitor-community/bluetooth-le: 7.1.0

peitschie, 2025-05-15:
Ah! Thanks for the report and the patch. I'll get a release together today.

closed 2025-05-16 by peitschie

peitschie, 2025-05-16:
A fix has been pushed out in 7.1.1

You've made this very simple to achieve 🙂

PR #763 (2025-05-16): fix(filters): restore filtering only with name functionality
Fixes #762

Essentially restores only name filtering after it was removed in the 7.0.0 -> 7.1.0 transition